tscircuit/docs-old #30: Digital Logic Tutorials

seveibar, 2024-10-21:


TSP06, 2024-10-23:
Hey @seveibar can you provide a detailed description for this issue

seveibar, 2024-11-01:
@TSP06 sorry for the delay, we basically want to provide some examples for digital logic so that course instructors use tscircuit for their diagrams.

Here's an example tutorial with many schematics we'd like to emulate. https://www.youtube.com/watch?v=lumpHzyakVs

[image]

I'm going to create sub-issues to make this more concrete!

TSP06, 2024-11-02:
So basically in first part I have to create a video tutorial demonstrating how to create and and or gate using ttl chips @seveibar

seveibar, 2024-11-02:
@TSP06 no video tutorial needed, we just need to create a markdown document (there should be a tutorials directory) an example on the website is here: https://docs.tscircuit.com/tutorials/blinking-led-circuit