tscircuit/docs-old #34: Create a tutorial that demonstrates how to create an AND and OR gate using TTL logic chips

seveibar, 2024-11-01:


seveibar, 2024-11-01:
/bounty $50

Srayash, 2024-11-02:
/attempt #34

Srayash, 2024-11-02:
/cancel #34

Niharika0104, 2024-11-03:
@seveibar I have added schematic circuits for and and or gates.Can you check and verify this?
https://stackblitz.com/edit/tscircuit-ttl-and-or-gates?file=lib%2FMyCircuit.manual-edits.ts

seveibar, 2024-11-04:
Hey @Niharika0104 that's pretty great work, but we need to do a couple things to make this a tutorial:

- [ ] Create the AND gate snippet on https://snippets.tscircuit.com using a TTL chip imported from JLCPCB
  - [ ] Use pushbuttons (e.g. @tsci/seveibar.push-button from the import menu) to allow setting each signal
  - [ ] Showcase each input and output symbol with an led
  - [ ] Ensure the 3d preview and PCB preview fully routes
- [ ] Write a markdown tutorial file in the docs repo (this repo) that explains how it was created with screenshots of the 3d preview and schematic preview.

Same for the OR gate. I can't say that this is an easy issue at the moment, but we do plan on creating a lot of issues like this as we go!

Niharika0104, 2024-11-05:
@seveibar [image]
I'm trying to add two examples, one for an OR gate and another for an AND gate, but I'm encountering an error that says I can't have multiple `manual-edits.tsx` files. It seems that a single project can only accommodate one circuit diagram. Am I correct in this assumption? Are you okay with me creating one Stackblitz project to demonstrate the AND gate and another project for the OR gate?

seveibar, 2024-11-05:
@Niharika0104 so stackblitz currently uses a really really old version of tscircuit, so to be clear stackblitz projects aren't acceptable generally for this task. The code has to run on https://snippets.tscircuit.com because the CLI that runs on stackblitz is very out of date.

seveibar, 2024-11-05:
Also note that that AND gate and OR gate would be separate snippets. This may be a difficult initial bounty.

Niharika0104, 2024-11-05:
@seveibar So you want me to demostrate the examples on https://snippets.tscircuit.com/  not on stackbliz??

seveibar, 2024-11-06:
@Niharika0104 I guess i don't really get your AND gate, shouldn't you have like two push buttons, and if someone presses BOTH of the push buttons a light turns on or something? You have a chip in there but there's no input to it.

Let's make it so...

- There are 2 push buttons
- There are 4 LEDs
- There's a USBC connector for power
- There's an AND logic chip (that is popular and in-stock, probably from texas instruments)
- When BUTTON1 is pressed, LED1 turns on
- When BUTTON2 is pressed, LED2 turns on
- When BUTTON1 and BUTTON2 is pressed, LED3 turns on because of the AND gate
- When there is power (USBC is connected) LED4 turns on
- The schematic is readable (not everything clumped)
- The PCB has no errors

I have reduced the scope of this issue and the bounty to only the AND gate

seveibar, 2024-11-06:
@Niharika0104 you may want to reference the [USBC Flashlight](https://snippets.tscircuit.com/editor?snippet_id=b210fe24-41d6-4465-b6a5-e99658e1dffa)

Niharika0104, 2024-11-06:
I will add them.Right now I'm using this IC7408 from jlcpcb(this is also an AND gate).
I will do the changes and get back to you.

Niharika0104, 2024-12-01:
/attempt #34

closed 2024-12-09 by imrishabh18