tscircuit/docs-old #35: Create a tutorial that demonstrates how to create an OR gate using TTL logic chips

seveibar, 2024-11-06:
Same criteria as #34, the hardest part is creating the snippet. The markdown doc is easy after

- There are 2 push buttons
- There are 4 LEDs
- There's a USBC connector for power
- There's an AND logic chip (that is popular and in-stock, probably from texas instruments)
- When BUTTON1 is pressed, LED1 turns on
- When BUTTON2 is pressed, LED2 turns on
- When BUTTON1 or BUTTON2 is pressed, LED3 turns on because of the OR gate
- When there is power (USBC is connected) LED4 turns on
- The schematic is readable (not everything clumped)
- The PCB has no errors

See [USBC Flashlight](https://snippets.tscircuit.com/editor?snippet_id=b210fe24-41d6-4465-b6a5-e99658e1dffa)

@ShiboSoftwareDev curious if you'd be interested in an issue like this

/bounty $50

ShiboSoftwareDev, 2024-11-06:
@seveibar If I understand correctly you want something like bun guides, create a snippet and then instructions of how to create and use it

seveibar, 2024-11-06:
Yep!!!

ShiboSoftwareDev, 2024-11-11:
/attempt #35

Abse2001, 2024-12-01:
/attempt #35

PR #41 (2024-12-03): added OR-Gate tutorial
/claim #35
/closes #35
closed 2024-12-03 by seveibar